ROBOTIS-GIT/DynamixelShield #10: Compiler warnings when you build for Arduino Due or Teensy or ...

KurtE, 2019-08-05:
I thought I would take a quick look at the shield code and setup.  I ordered one which should arrive in the next few days...  

I tried compiling for a few different platforms.  

In particular I have tried the scan_dynamixel example. 

And I receive the warning/messages: 
```
C:\Users\kurte\Documents\Arduino\libraries\DynamixelShield\src\util\RobotisRemoteController.cpp:29:202: note: #pragma message: 

Warning : You CAN NOT use SoftwareSerial version of the RobotisRemoteController function(only HardwareSerial version enabled), because this board DOES NOT SUPPORT SoftwareSerial.h

 #pragma message("\r\nWarning : You CAN NOT use SoftwareSerial version of the RobotisRemoteController function(only HardwareSerial version enabled), because this board DOES NOT SUPPORT SoftwareSerial.h")

                                                                                                                                                                                                          ^

C:\Users\kurte\Documents\Arduino\libraries\DynamixelShield\src\util\RobotisRemoteController.cpp:30:144: note: #pragma message: 
```
Note: This example does not use the RobitisRemoteController: it is included by the main header file.  And it will print this regardless if you include this or not, if it thinks for some reason your program has included SoftwareSerial.h...  

Also side note, when compile for Due there is another warning message, abut exit... But I think that is outside of your stuff...

Will have more questions/comments later, like what is the relationship of this versus DynamixelSDK verus DynamixelWorkbench, and is there one solution that works across platforms?   That is if I have a hexapod with 18 servos, that I may want to run using OpenCM or OpenCR or maybe Arduino Mega or maybe Teensy... Can this be done by one set of code.

Should the above be discussed on github issues (which project) or Robotis Forum or RobotSource?

Kurt

OpusK, 2019-08-06:
Hi, @KurtE 

> This example does not use the RobitisRemoteController: it is included by the main header file. And it will print this regardless if you include this or not, if it thinks for some reason your program has included SoftwareSerial.h...

This is because RobitisRemoteController uses SoftwareSerial by default, so if the library does not exist, it is put to inform the user.

> Will have more questions/comments later, like what is the relationship of this versus DynamixelSDK verus DynamixelWorkbench, and is there one solution that works across platforms? That is if I have [...]

As you know, SDK and Workbench support only OpenCR, OpenCM 9.04.
I think [Dynamixel2Arduino](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino) could be the solution you want.

KurtE, 2019-08-06:
Hi @OpusK,
To, me the warning message is sort of out of place as I am not trying to use that class at all.., so as the user the error message makes no sense, in the context of the example program for scanning servos.

Yes as you mentioned I understand that the SDK and workbench are only supported by OpenCM and OpenCR.   And then there is this library DynamixelShield, which I have seen is now derived from Derived from Dynamixel2Arduino.   So far I have not seen any documentation on this one.  Like the DynamixelShield mentions a limit of 16 servos, does Dynamixel2Arduino have the same restriction?   I guess next up is to look through the code to get an idea of how it is setup... 

But I still don't understand the reasons for having so many similar but divergent libraries for doing a lot of the same things.  I sort of understand maybe the need for Workbench versus SDK, but wish there was one library that supported most all platforms.   

Of course I do have the option to continue to use my own libraries, such as: https://github.com/KurtE/BioloidSerial
But would like to migrate some of the programs over to "standard" libraries.

OpusK, 2019-08-07:
@KurtE 

> So far I have not seen any documentation on this one. Like the DynamixelShield mentions a limit of 16 servos, does Dynamixel2Arduino have the same restriction?

As explained in the DynamixelShield manual, the default limit of 16 servos is correct for hardware current limiting.
However, this is only limited by assuming a continuous connection in daisy chain. Therefore, you can change it in the [config file.](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/config.h#L41)

> But I still don't understand the reasons for having so many similar but divergent libraries for doing a lot of the same things. I sort of understand maybe the need for Workbench versus SDK, but wish there was one library that supported most all platforms.

As I said before, the SDK and Workbench are designed for PC software.
The reason for supporting OpenCM9.04 and OpenCR is that there was no suitable solution at the time (I was at another company at the time).

When I updated DynamixelShield, I needed a library that would be common to Arduino, and I created Dynamixel2Arduino.
I tried to use the workbench or SDK, but the Workbench(It uses the SDK internally)'s `string` and `dynamic allocation` methods were not suitable for low memory boards like UNO and Mega2560.
And, because PC users prefer this `String` method and `dynamic allocation`, I could not ask for their modification.

Anyway, I fully agree with you.
To address this, the RobotSource or ROBOTIS forum that we have previously shared seems to be appropriate.

KurtE, 2019-08-07:
@OpusK ,

Right now playing with migrating my Hexapod code base, that I had ported over to OpenCM/CR over to this and see how well it does.  

I always have minor questions on why some things like setProtocol takes a floating point value instead of some integer as it is either 1 or 2...  But that is minor.

Trying to figure out things like: Currently I have the code setup that on OpenCM boards, the servos might be on the on Board Serial port and/or might be on the open cm 485 expansion board, so I have protocol 1 and protocol 2 objects, as well as two port objects, and then I try to find the hexapods servos in all of the possible configurations... I take it with this library, I just need to switch protocol to go between 1 and 2, but I am guessing I would need two of the Dynamixel2Arduino objects to handle both ports?  

I totally understand your desire to remove all of the mallocs and frees and ... In fact maybe I recommended that sometime in the past ;) Although at times I don't have issues of malloc/new once at startup type configurations...  

The question also is, suppose I wish to support this on some other processors as well.  Example the Teensy 4, that was released today: https://www.pjrc.com/teensy-4-0/
As well as some of the other Teensy processors.   With many of my boards, I have setup to use a TTL level shifting chip to convert to 5v... Like a TC7WT141FU, so can easily just pass in Serial port and a R/W select pin for this.   However they also have the ability to configure the UART into half duplex mode (TX pin), and you then need to switch a register setting to configure if you are doing input or output.... Wonder how hard it would be to allow some form of call out or??? 

As for the bioloidSerial library (original version was done for Trossen AVR boards ...) But yes this version is Arduino specific. 
I do have a version I did for linux that is up on my Raspberry Pi project (https://github.com/KurtE/Raspberry_Pi) where the BioloidEX directory is a port of the earlier code, that then runs on RPI, or Odroid, or Edison, or ...  However I don't think I migrated any support yet into doing protocol 2 there...

OpusK, 2019-08-08:
@KurtE 

> I always have minor questions on why some things like setProtocol takes a floating point value instead of some integer as it is either 1 or 2... But that is minor.

The SDK and Workbench use this way, so I applied it as well.
Presumably, the e-manual is represented by 1.0 and 2.0, and the protocol developer may have done so, considering minor protocol changes like 2.1 later.
This is simply my personal guess.

> Trying to figure out things like: Currently I have the code setup that on OpenCM boards, the servos might be on the on Board Serial port and/or might be on the open cm 485 expansion board, so I have [...]

Currently, only one port is used on the side that handles packets.
I will update this.
If this is updated, you can create two Dynamixel2Arduino objects if the DXL hardware is isolated, such as OpenCM9.04.

> The question also is, suppose I wish to support this on some other processors as well. Example the Teensy 4, that was released today: https://www.pjrc.com/teensy-4-0/
As well as some of the other Teensy processors. With many of my boards, I have setup to use a TTL level shifting chip to convert to 5v... Like a TC7WT141FU, so can easily just pass in Serial port and a R/W select pin for this. However they also have the ability to configure the UART into half duplex mode (TX pin), and you then need to switch a register setting to configure if you are doing input or output.... Wonder how hard it would be to allow some form of call out or???

> As for the bioloidSerial library (original version was done for Trossen AVR boards ...) But yes this version is Arduino specific.
I do have a version I did for linux that is up on my Raspberry Pi project (https://github.com/KurtE/Raspberry_Pi) where the BioloidEX directory is a port of the earlier code, that then runs on RPI, or Odroid, or Edison, or ... However I don't think I migrated any support yet into doing protocol 2 there...

The best thing is that the SDK will be available on all platforms.
However, this doesn't seem to be something we can easily decide here.
Rather, it would be easier to work with generalizing low-spec libraries such as Dynamixel2Arduino to make them available on all platforms.
For example, if we develop such a general library, Dynamixel2Arduino will use it.

If you are willing to participate in the development of this generalized library, I will ask my boss about the project.

OpusK, 2019-08-08:
@KurtE 

I've updated to support multiple ports.
We haven't released it yet, please use the code below.

- ROBOTIS-GIT/Dynamixel2Arduino#15

KurtE, 2019-08-08:
@OpusK ,

looks interesting,  not fully sure yet how to use it to do multiple ports.   

My guess is you create the Dynamixel2Arduino object with the information for one of the ports, and you create another DYNAMIXEL::SerialPortHandler object for the other port.  You can then call setPort with the new one to use it, but currently I don't see any method to retrieve the current one (like the one it created internally...).  Maybe having a getPort() method would allow that.  

Also once you have all of this in place, will need to figure out some of the relationships.  Like when I do ping operation, do I do it several times one for each port and one for each protocol...  Are the lists then combined... Probably... Will probably need to experiment.

OpusK, 2019-08-09:
@KurtE,

> My guess is you create the Dynamixel2Arduino object with the information for one of the ports, and you create another DYNAMIXEL::SerialPortHandler object for the other port. You can then call [...]

You can use the way you say it, but I recommend creating two Dynamixel2Arduino objects. (Unless the system is low on memory)

OpusK, 2019-08-09:
@KurtE, 

I released version [0.1.5](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/releases/tag/0.1.5) with regard to Multiple port support.
In a few hours, it will be updated on the Arduino Library Manager.

KurtE, 2019-08-09:
Sounds good, I will try converting some more of the code to it and see how hard it is to convert from DynamixelSDK code to it. 

 

Thanks again

OpusK, 2019-09-16:
Hi, @KurtE 

If you have any problems or improvements in use, please let me know.
And for the compilation warning issue you raised, I'll remove the code that triggers the warning and add it to the manual.

KurtE, 2019-09-16:
@OpusK - Will let you know. 

I have been doing other diversions with the new Teensy for a bit, but need/want to get back to Robotics stuff, including working on this. 

Also need to bet back to Turtlebot stuff and figure out the state of ROS2 stuff and get on track again. 

But there are just too many fun distractions :D

OpusK, 2019-09-20:
Hi, @KurtE 

This issue has been fixed in 0.3.0.

OpusK, 2019-09-24:
@KurtE 

May I close this issue?

KurtE, 2019-09-24:
Yes

closed 2019-09-24 by KurtE