ROBOTIS-GIT/DynamixelShield #12: question using dynamixel shield

techjjun, 2019-11-14:
I'm using dynamixel shield and arduino uno board to control XL-430 moror.

and running example file in arudino ide,,   Serial.print ()  line doesn't work but motor is working. 

it just keep printing ????  in serial monitor..  

I set port baudrate to 57600 for dynamixel, and 57600 for UART port.

is it not possible to communicate in Serial??

OpusK, 2019-11-15:
Hi, @techjjun 

In the [e-manual](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#layout), 0,1 (RX, TX) pins are used for DYNAMIXEL(DXL) communication.
This is happening in UNO and Mega because the pins are connected to USB (Serial).

Therefore, DXL communication and serial cannot be used at the same time.
An alternative method is to use SoftwareSerial to print to another pin. In case of Mega, you can use another serial like Serial1.

techjjun, 2019-11-15:
Thanks for the reply!  I have some more question.

currently I want to control XL-430 motor with shield that can change velocity or direction depending on some value, and this value comes from PC. 

So because of hardware serial is used for the motor, how can i communicate PC with shield?, 
sending a value from PC to shield, do i have to use bluetooth communication using 7,8 pin as SoftwareSerial??

OpusK, 2019-11-18:
@techjjun 

> So because of hardware serial is used for the motor, how can i communicate PC with shield?,
sending a value from PC to shield, do i have to use bluetooth communication using 7,8 pin as SoftwareSerial??

If you have to use UNO, it looks like you have to use a USBtoSerial device, whether Bluetooth or not.

If you have a board like Leonardo, you can use it like this:
- PC --- USB port --- Leonardo Board --- Serial1(0,1 pin) --- DYNAMIXEL Shield

OpusK, 2019-12-23:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-12-23 by OpusK

JamesNewton, 2020-07-30:
For step by step directions and parts list as well as code for setting baud and ID, see:
https://github.com/JamesNewton/Dynamixel-Shield-Setup