ROBOTIS-GIT/DynamixelShield #13: Problem with Arduino Due

3sigma, 2019-12-19:
Hello,

In order to support Arduino Due, this line:
[https://github.com/ROBOTIS-GIT/DynamixelShield/blob/7712fd31fed1da681aa4bc9d8f4f7a0dc01f1313/src/DynamixelShield.h#L31](https://github.com/ROBOTIS-GIT/DynamixelShield/blob/7712fd31fed1da681aa4bc9d8f4f7a0dc01f1313/src/DynamixelShield.h#L31)

should be:
`#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560) || defined(ARDUINO_SAM_DUE)`

Thanks in advance,

Nicolas

OpusK, 2019-12-20:
Hi, @3sigma 

Thank you for your bug report.
We will fix it in a future release.

OpusK, 2019-12-23:
@3sigma 

I have released [0.1.5](https://github.com/ROBOTIS-GIT/DynamixelShield/releases/tag/0.1.5)
At the end of the Arduino update cycle, it will be available in the library manager. (This may take up to several hours.)

closed 2020-02-07 by OpusK