ROBOTIS-GIT/DynamixelShield #22: DynamixelShield and RX-64

guilhermeaspereira, 2020-08-20:
Hello,

I would like to control the RX-64 servo with the Shield. Currently, it is not working. Is there anything we can do?

Thanks,
Guilherme

ROBOTIS-Will, 2020-08-21:
Hi Guilherme,

In order to enable RX series, you should modify the Dynamixel2Arduino library [config.h](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/config.h#L7) file in your local disk as below.
```
#define ENABLE_ACTUATOR_RX              1
```
If you cannot find the library files, please use [Manual installation](https://www.arduino.cc/en/guide/libraries#toc5) method instead of using the Library Manager.
Thank you.

ROBOTIS-Will, 2020-08-25:
Hi, which Arduino board do you use with the DYNAMIXEL Shield?
Please make sure that you have set the [power jumper](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#connecting-power) properly, as well as the [switch](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#switches).
Thank you.

guilhermeaspereira, 2020-08-28:
Hi Will,

I'm using Arduino UNO. Yes, I tested different power methods and also changed the switch. Unfortunately, it is still not working. Do you suggest a simple test? I'm currently using some of the examples: led, position control, etc.

Thanks,
Guilherme

ROBOTIS-Will, 2020-09-01:
@guilhermeaspereira 
Hi Guilherme,
Thank you for your patience. We've tested and confirmed that AX and RX series have difficulty in communicating with Arduino Uno at Baudrate 57600bps due to margin of error. In this case, please use 9600bps or 115200bps instead.
Thank you.

guilhermeaspereira, 2020-09-01:
Hi Will,

This works! Thanks for your help!

Guilherme

closed 2020-09-02 by ROBOTIS-Will