ROBOTIS-GIT/DynamixelShield #23: Sending Bytes through LN-101/Serial-USB Converter

grndctrl2majtom, 2020-11-12:
Do you have any examples on how to write a user defined angle to the dynamixel shield using the LN-101?  I am having some issues with DEBUG_SERIAL.available, DEBUG_SERIAL.read functions.

ROBOTIS-Will, 2020-11-16:
Hi @grndctrl2majtom 
There are some examples in the default library that use available() and read().
One of them is the [factory_reset](https://github.com/ROBOTIS-GIT/DynamixelShield/blob/master/examples/dynamixel_protocol/factory_reset/factory_reset.ino) and you can refer from [line 62](https://github.com/ROBOTIS-GIT/DynamixelShield/blob/master/examples/dynamixel_protocol/factory_reset/factory_reset.ino#L62).
In order to receive multiple letters, you might want to use string type of variable and something like readString() to read multiple characters.
You can convert the string variable to a float to use for the setGoalPosition() of [DYNAMIXEL Shield API](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#dynamixel2arduino-class).

ROBOTIS-Will, 2020-12-21:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-21 by ROBOTIS-Will