ROBOTIS-GIT/DynamixelShield #25: Writing PID Gains to MX64

grndctrl2majtom, 2020-11-22:
Hello, I'm having some issues writing PID commands to the MX64. It does not appear to actually be assigning the gains to the motor. Here is how I am doing it.

dxl.writeControlTableItem(POSITION_P_GAIN,1,254);
  dxl.writeControlTableItem(POSITION_D_GAIN,1,50);
  dxl.writeControlTableItem(POSITION_I_GAIN,1,200);

ROBOTIS-Will, 2020-11-23:
Hi, thank you for the update.
The read / writeControlTableItem functions can actually be located in the setup function depending on your software design.
And these are located in the RAM area, so you should set these values every time system is turned on.
Thanks!

ROBOTIS-Will, 2020-12-21:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-21 by ROBOTIS-Will