ROBOTIS-GIT/DynamixelShield #31: AX12

wngfra, 2021-06-04:


wngfra, 2021-06-04:
Hardware: Arduino Portenta H7 + Dynamixel MKR Shield
Using 11.1v 3S Li-Po as power source, V and G measure correct on the TTL connector but running the example `scan_dynamixel` found no servos. The Dynamixel Protocol is set to 1.0.
Any idea what the potential problem is?

wngfra, 2021-06-05:
Tested with MKR Wifi 1010 and the Dynamixel Shield works perfectly. So the problem is very likely to be specific with Portenta H7, either my Portenta board is broken or there is a different pin setup in the codes.

ROBOTIS-Will, 2021-06-07:
Hi @wngfra 

It is reported that the Portenta triggers TX -> RX switching (TX_EN, A6) too early which causes truncated TX signal.
We're seeking for a solution for this at the moment.
[image]

adityakamath, 2022-04-27:
Hi @ROBOTIS-Will is there any progress on this issue? I have the same problem.

ROBOTIS-Will, 2022-04-29:
Hi @adityakamath 
Thanks for reminding me of this issue.
Until the TX status monitoring feature like tx_is_active() is implmented to the mbed library, [this DYNAMIXEL2Arduino update](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/commit/2800ae9dacb6949ddacff65131ab100a1af0b26b) may resolve the issue.
Related issue thread : https://github.com/arduino/ArduinoCore-mbed/issues/389
Please let me know if there's any issue with this update.
This update will be included in the next release.

Thanks!

adityakamath, 2022-04-29:
Thanks @ROBOTIS-Will, I can confirm that this update works for me. Tested the scan_dynamixel and position_mode examples with XL330 and XL430.

closed 2025-04-21 by jonghee-son