ROBOTIS-GIT/DynamixelShield #33: RX-28 is not detected

Jayadev22, 2021-11-13:
Servo model: RX-28
Setup: DynamixelShield on ArduinoUNO. Debug serial using another ArduinoUNO as UART-Serial bridge.
Situation: I recently purchased the DynamixelShield for a project to control a couple of RX-28s that I had from another project. This decision came after I was unable to even detect the motors using a self-made RS-485 to USB converter, though this same setup worked on that earlier project. I assumed that the converter is broken as the cables were pretty worn-out and bought the shield. Then when I tried to scan one of the motors using scan_dynamixel example code, nothing showed up. The power supply is set to 14V and the even the motor blinks once when powered on, which is normal. But after that, nothing works nor there are any LED indications of a particular error. Just as a control experiment, I connected an AX-12 that I borrowed from another robot, which got clearly detected with the same scan_dynamixel example code. It’s pretty evident that the issue is with the RX-28 motors. I’m not sure how to debug this to get them working as they are not even discoverable to maybe factory-reset them or something. Any help from the from the community will be appreciated, thank you.

ROBOTIS-Will, 2021-11-15:
Hi @Jayadev22 

The DYNAMIXEL2Arduino library deactivates some discontinued DYNAMIXEL series in order to save the memory space in some boards.
In order to enable the RX-28, please set the `ENABLE_ACTUTATOR_RX` to `1` in the setup below.
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/config.h#L7

Thanks

Jayadev22, 2021-11-15:
Hi @ROBOTIS-Will 

Thank you for replying. I tried what you suggested, but unfortunately it didn't make any difference. The RX-28 is still undetectable. Any other possibilities that you can think of.? I would readily try whatever is possible to get them running.

Thank you again.

ROBOTIS-Will, 2021-11-16:
If you have changed the buad rate of the RX-28, the scan_dynamixel example may not detect it as the example only detects as below.
```cpp
const int32_t buad[MAX_BAUD] = {57600, 115200, 1000000, 2000000, 3000000};
```

If you have USB2DYNAMIXEL or U2D2 interface, I'd recommend recovering the firmware with [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/rplus1/dynamixel_wizard/)

Thank you.

closed 2025-04-21 by jonghee-son