ROBOTIS-GIT/DynamixelShield #34: MX-64AR with Arduino UNO (Dynamixel Shield)

KoutarouTakahashi, 2021-11-25:
お世話になっております。MX-64ARをShieldで動かすために購入したのですが、シリアルモニタの表示がうまく行かず困っています。e-manualを読んでライブラリをインクルードした上でサンプルのscan_dynamixelとidを動かしました。しかし、文字化けがひどかったのでArduinoの0→7,1→8ピンをジャンパ線で接続したところscan_dynamixelは
SCAN PROTOCOL 1
SCAN BAUDRATE 57600
と出るようになりましたがコードの他の部分に書いてある文章が出てきません。同じ条件でidの方をやってみると何も応答しません。
マイコンはArduino UNOで、12V,2Aを印加しました。日本支社の方からこちらにコードを乗せるよう言われたためこちらで質問させていただきます。よろしくおねがいします。(サンプルコードのidとscan Dynamixelを入れたつもりですが、載せます。)

I bought Dynamixel Shield to control MX-64AR but Serial Monitor can't show up well.  So, I read e-manual ,included library and run sample code "id" and "scan dynamixel",however,Garbled characters were shown. Then, I connected 0pin to 7pin and 1pin to 8pin on Arduino port with jumper, and serial monitor responsed as follow (sample: scan dynamixel)

SCAN PROTOCOL 1
SCAN BAUDRATE 57600

but the other sentence wouldn't show up.. At same condition, tried "id",but monitor responsed nothing.
Using micro computer is Arduino UNO.  When I tried the shield, The power supply is 12V voltage and 2A current.
 ROBOTIS Japan branch Technical Department team asked me to write down this question on github. In this situation, I want to ask the solution.  Please answer my question.

code is follow(scan dynamixel and id, both of them are contained in Library Dynamixel Shield basic tags)
scan dynamixel

#include <DynamixelShield.h>

#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560)
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DEBUG_SERIAL soft_serial
#elif defined(ARDUINO_SAM_DUE) || defined(ARDUINO_SAM_ZERO)
  #define DEBUG_SERIAL SerialUSB
#else
  #define DEBUG_SERIAL Serial
#endif

#define MAX_BAUD  5
const int32_t buad[MAX_BAUD] = {57600, 115200, 1000000, 2000000, 3000000};

Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN);

//This namespace is required to use Control table item names
using namespace ControlTableItem;

void setup() {
  // put your setup code here, to run once:
  int8_t index = 0;
  int8_t found_dynamixel = 0;

  // For Uno, Nano, Mini, and Mega, use UART port of DYNAMIXEL Shield to debug.
  DEBUG_SERIAL.begin(115200);   //set debugging port baudrate to 115200bps
  while(!DEBUG_SERIAL);         //Wait until the serial port is opened
    
  for(int8_t protocol = 1; protocol < 3; protocol++) {
    // Set Port Protocol Version. This has to match with DYNAMIXEL protocol version.
    dxl.setPortProtocolVersion((float)protocol);
    DEBUG_SERIAL.print("SCAN PROTOCOL ");
    DEBUG_SERIAL.println(protocol);
    
    for(index = 0; index < MAX_BAUD; index++) {
      // Set Port baudrate.
      DEBUG_SERIAL.print("SCAN BAUDRATE ");
      DEBUG_SERIAL.println(buad[index]);
      dxl.begin(buad[index]);
      for(int id = 0; id < DXL_BROADCAST_ID; id++) {
        //iterate until all ID in each buadrate is scanned.
        if(dxl.ping(id)) {
          DEBUG_SERIAL.print("ID : ");
          DEBUG_SERIAL.print(id);
          DEBUG_SERIAL.print(", Model Number: ");
          DEBUG_SERIAL.println(dxl.getModelNumber(id));
          found_dynamixel++;
        }
      }
    }
  }
  
  DEBUG_SERIAL.print("Total ");
  DEBUG_SERIAL.print(found_dynamixel);
  DEBUG_SERIAL.println(" DYNAMIXEL(s) found!");
}

void loop() {
  // put your main code here, to run repeatedly:
}

code id

#include <DynamixelShield.h>

#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560)
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DEBUG_SERIAL soft_serial
#elif defined(ARDUINO_SAM_DUE) || defined(ARDUINO_SAM_ZERO)
  #define DEBUG_SERIAL SerialUSB    
#else
  #define DEBUG_SERIAL Serial
#endif

const uint8_t DEFAULT_DXL_ID = 1;
const float DXL_PROTOCOL_VERSION = 2.0;

DynamixelShield dxl;

//This namespace is required to use Control table item names
using namespace ControlTableItem;

void setup() {
  // put your setup code here, to run once:
  uint8_t present_id = DEFAULT_DXL_ID;
  uint8_t new_id = 0;
  
  // For Uno, Nano, Mini, and Mega, use UART port of DYNAMIXEL Shield to debug.
  DEBUG_SERIAL.begin(115200);
  while(!DEBUG_SERIAL);
  
  // Set Port baudrate to 57600bps. This has to match with DYNAMIXEL baudrate.
  dxl.begin(57600);
  // Set Port Protocol Version. This has to match with DYNAMIXEL protocol version.
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);

  DEBUG_SERIAL.print("PROTOCOL ");
  DEBUG_SERIAL.print(DXL_PROTOCOL_VERSION, 1);
  DEBUG_SERIAL.print(", ID ");
  DEBUG_SERIAL.print(present_id);
  DEBUG_SERIAL.print(": ");
  if(dxl.ping(present_id) == true) {
    DEBUG_SERIAL.print("ping succeeded!");
    DEBUG_SERIAL.print(", Model Number: ");
    DEBUG_SERIAL.println(dxl.getModelNumber(present_id));
    
    // Turn off torque when configuring items in EEPROM area
    dxl.torqueOff(present_id);
    
    // set a new ID for DYNAMIXEL. Do not use ID 200
    new_id = 100;
    if(dxl.setID(present_id, new_id) == true){
      present_id = new_id;
      DEBUG_SERIAL.print("ID has been successfully changed to ");
      DEBUG_SERIAL.println(new_id);

      new_id = DEFAULT_DXL_ID;
      if(dxl.setID(present_id, new_id) == true){
        present_id = new_id;
        DEBUG_SERIAL.print("ID has been successfully changed back to Original ID ");
        DEBUG_SERIAL.println(new_id);
      }else{
        DEBUG_SERIAL.print("Failed to change ID to ");
        DEBUG_SERIAL.println(new_id);
      }
    }else{
      DEBUG_SERIAL.print("Failed to change ID to ");
      DEBUG_SERIAL.println(new_id);
    }
  }
  else{
    DEBUG_SERIAL.println("ping failed!");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}

ROBOTIS-Will, 2021-11-25:
Hi @KoutarouTakahashi 

Thank you for your inquiry and detailed explanation about the issue.
If you are using Arduino Uno, it is sharing one serial port with USB and Tx/Rx port.

DYNAMIXEL Shield use Tx/Rx port to communicate with DYNAMIXEL, therefore, Arduino terminal will not work properly.

Please refer to the issue thread #18 for more details.

Thank you.

KoutarouTakahashi, 2021-11-25:
Hi ,Mr.Will.
Thanks for answering.
I read #18 and learned one  UNO can't show data on serial monitor because of one USB port ,RX/TX port collision.
So, I noticed that how about using LCD (Liquid Crystal Display) instead of serial monitor?
I think it can show up data by using I2C communication. and don't have to concern about port collision.

Thank you.

ROBOTIS-Will, 2021-11-29:
Hi @KoutarouTakahashi 

Yes, as long as pins you are going to use does not conflict with [DYNAMIXEL Shield pins](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#layout), you may freely use the shield.
Thank you.

KoutarouTakahashi, 2021-12-03:
Hi @ROBOTIS-Will 
Thanks to your advice, I  managed to control MX64-AR with lcd!
thank you very much!

closed 2025-04-21 by jonghee-son