ROBOTIS-GIT/DynamixelShield #39: How can I sync write the AX-12A servos to the goal position?

chenxinencode, 2022-04-25:
The sync write function seems to only support protocool 2.0, but the AX-12A only support protocool 1.0.

ROBOTIS-David, 2022-05-12:
Sorry for late replay, 

DYNAMIXEL Protocol 1.0 products such as Ax series support Sync Write instruction. See [DYNAMIXEL Protocol 1.0](https://emanual.robotis.com/docs/en/dxl/protocol1/#sync-write) page

For further inquiries, I highly recommend visiting our [community page](https://community.robotis.us/c/community/5) and find information you need or raise ticket to ask. Our experts and power users are activly playing around at that place for 24 hours.

ROBOTIS-David, 2022-07-19:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2022-07-19 by ROBOTIS-David